pythops/jetson-image #155: How to determine Nano revision?

parkerlreed, 2025-04-12:
```
revision: the revision number. Only required for jetson-nano board. The possible values are: 100, 200 or 300.
```

I seems to have A02 (2GB RAM). Which one does this match with?

pythops, 2025-04-12:
in your case, you should use `jetson-nano-2gb` as board
```
just build-jetson-image -b jetson-nano-2gb
```
the revision only applies to jetson nano whcih are 4gb

closed 2025-04-12 by pythops

parkerlreed, 2025-04-12:
Quick question. I downloaded 22.04 jammy. Added `cuda` to the packages listing. Ran the build-jetson-image and it fails at installing cuda since the repo that was added seems to have an invalid replacement

```
Err:6 https://repo.download.nvidia.com/jetson/<SOC> r32.7 Release
  404  Not Found [IP: 23.53.11.134 443]
```

Do I need to set this somewhere?

pythops, 2025-04-12:
can you list here the exact steps you took ?

parkerlreed, 2025-04-12:
```
just build-jetson-rootfs 22.04
# add cuda to l4t packages file
just build-jetson-image -b jetson-nano-2gb -l 32
```

pythops, 2025-04-12:
maybe try `cuda-toolkit` instead of `cuda`

parkerlreed, 2025-04-12:
Same deal, failing with the 404 on the invalid URL

```
Setting up nvidia-l4t-kernel-dtbs (4.9.337-tegra-32.7.5-20240611161210) ...
Pre-installing kernel-dtbs package, skip flashing
Setting up nvidia-l4t-kernel-headers (4.9.337-tegra-32.7.5-20240611161210) ...
Setting up nvidia-l4t-camera (32.7.5-20240611161210) ...
Setting up nvidia-l4t-gstreamer (32.7.5-20240611161210) ...
Processing triggers for nvidia-l4t-kernel (4.9.337-tegra-32.7.5-20240611161210) ...
Processing triggers for libc-bin (2.35-0ubuntu3.9) ...
/build/Linux_for_Tegra
Removing QEMU binary from rootfs
Removing stashed Debian packages from rootfs
L4T BSP package installation completed!
Rename ubuntu.desktop --> ux-ubuntu.desktop
Disabling NetworkManager-wait-online.service
Disable the ondemand service by changing the runlevels to 'K'
Success!
--> 63bfebd1197b
STEP 14/23: RUN chroot /build/Linux_for_Tegra/rootfs/ /bin/bash -c "chown -R jetson:jetson /home/jetson && chmod 1777 /tmp"
--> c7eeff1676ec
STEP 15/23: RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev &&     chroot rootfs /bin/bash -c "echo 'nameserver 8.8.8.8' > /etc/resolv.conf && apt update; apt install -y cuda-toolkit-10-2 $L4T_PACKAGES"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 https://repo.download.nvidia.com/jetson/common r32.7 InRelease [2555 B]
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Ign:4 https://repo.download.nvidia.com/jetson/<SOC> r32.7 InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Err:6 https://repo.download.nvidia.com/jetson/<SOC> r32.7 Release
  404  Not Found [IP: 23.53.11.134 443]
Hit:7 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Hit:8 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:9 https://repo.download.nvidia.com/jetson/common r32.7/main arm64 Packages [16.6 kB]
Reading package lists...
E: The repository 'https://repo.download.nvidia.com/jetson/<SOC> r32.7 Release' does not have a Release file.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package cuda-toolkit
Error: building at STEP "RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev &&     chroot rootfs /bin/bash -c "echo 'nameserver 8.8.8.8' > /etc/resolv.conf && apt update; apt install -y cuda-toolkit-10-2 $L4T_PACKAGES"": while running runtime: exit status 100
```

pythops, 2025-04-12:
`cuda-toolkit` is already included so you don't need to add anything actually.

parkerlreed, 2025-04-12:
Okay yeah seems to be working specifying nothing in the extra packages. I'll check if this build and can be flashed, thanks.